bigtreetech/BIGTREETECH-OCTOPUS-Pro #7: Octopus PRO dual Y axis

luca920, 2022-02-03:
I am trying to set the board to have dual Y axis, but am struggling in understanding how. 
Which config lines should I modifiy?

ebraiman, 2022-02-04:
Look in pins file for marlin for pins_BTT_OCTOPUS_V1_common.h  You might need to rename something like ...

#define Z_STEP_PIN                          PF11  // MOTOR 2
#define Z_DIR_PIN                           PG3
#define Z_ENABLE_PIN                        PG5
#ifndef Z_CS_PIN
  #define Z_CS_PIN                          PC6
#endif

#define Z2_STEP_PIN                         PG4   // MOTOR 3
#define Z2_DIR_PIN                          PC1
#define Z2_ENABLE_PIN                       PA0
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                         PC7
#endif

To Y2.

luca920, 2022-02-04:
Hi @ebraiman,
thank you for the answer. I will try.

just to be clear you mean that I should change the block (in this ase I think the one I like) like so:

#define Y2_STEP_PIN PG4 // MOTOR 3
#define Y2_DIR_PIN PC1
#define Y2_ENABLE_PIN PA0
#ifndef Y2_CS_PIN
#define Y2_CS_PIN PC7
#endif

is that right?

ebraiman, 2022-02-04:
I'll be Streaming tonight @ 6PM EST and I'll look at your issue.  The Channel is called "Ed's 3d Tech" on YouTube.  It will also be in livestream playlist if you miss it.

ebraiman, 2022-02-05:
Solution is about halfway thru the stream ... https://youtu.be/sW3z1BYNOPY

luca920, 2022-02-05:
You are really good!
I am follwing you (unfortunately was not able to get on live stream), but you have been really helpful!

I thank you again! (will open another thred btw on how to connect external driver for which I am still struggling too XD)