bigtreetech/BIGTREETECH-OCTOPUS-Pro #12: using vscode to flash firmware does not work

Majestix1970, 2022-05-26:
Hi,
 I am wondering if it is possible to flash the firmware on my Octopus Pro V1.0 with vscode/platformio using the USB-C connection.

this is my platformio.ini setting:
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = BIGTREE_OCTOPUS_PRO_V1_F429_USB

in plotformio device list the board shows up as COM12.

building the Marlin firmware and flashing it using the SD card works fin, but as soon as I want to directly flash it using thw USB-C connection I get the following error:
Uploading .pio\build\BIGTREE_OCTOPUS_PRO_V1_F429_USB\firmware.bin
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

embedded:startup.tcl:26: Error: Can't find board/stm32f429.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
*** [upload] Error 1
===================================================== [FAILED] Took 13.39 seconds =====================================================
Environment                      Status    Duration
-------------------------------  --------  ------------
BIGTREE_OCTOPUS_PRO_V1_F429_USB  FAILED    00:00:13.391
================================================ 1 failed, 0 succeeded in 00:00:13.391 ================================================

What do I do wrong? Or is the board not able to be flashed usinf the USB-C port (hope not...)?

Best regards
Majestix

yzheka, 2023-04-12:
this is deffinetely should be possible because I can update klipper firmware using `make flash FLASH_DEVICE=/dev/serial/by-id/...` on the board which is connected by usb to the Rasberry PI

podvlada, 2023-08-01:
Any update? I'm having exactly the same issue, same board. Thanks!