bigtreetech/BIGTREETECH-OCTOPUS-Pro #13: 2nd Tactile button on v1.0.1

Budavaril, 2022-05-29:
Can not find any description for a new 1.0.1 pro board (429) where is explained the function of that tactile button switch next to the SD slot and aligned with a wifi card black connector.
[image]

GadgetAngel, 2022-05-30:
> Can not find any description for a new 1.0.1 pro board (429) where is explained the function of that tactile button switch next to the SD slot and aligned with a wifi card black connector. [image]

It appears to be the software reset switch that they left off the original version of the pro board:

The schematic shows it but on the original board it was not there, so it looks like they fixed it with the new boards

[image]

Budavaril, 2022-05-30:
I saw this button on the schematic diagram (sheet5 bottom left),  but I believe this is the "normal" reset button what is physically on the top right corner of the board
[image]
.. 
I couldn't see second button on the shematic even I checked 3times..