bigtreetech/BIGTREETECH-OCTOPUS-Pro #14: Speaker pin?

jvandervyver, 2022-05-30:
Hi

What is the pin to which "SPK1" is connected?
If it isn't connected why include a speaker onboard?

Budavaril, 2022-05-30:
Hi,
Not for controlled by MCU as I guess you thought, so not connected for it. 
But it's connected to a HT7550-2 (SOT23-5) chip to watch the 5V rail voltage is present. If not, then will beeping. 

[image]

[image]

jvandervyver, 2022-05-30:
Wow I really didn't expect such a detailed response.  Thanks for explaining that explains why I hear the beeper when I power down.

> Not for controlled by MCU as I guess you thought

I did

closed 2022-05-30 by jvandervyver