bigtreetech/BIGTREETECH-OCTOPUS-Pro #17: PE11/GND reversed in Octopus Pro color PIN diagram

ph1ltep, 2022-07-05:
The PE11/GND pins are reversed in the color version of the pin diagram for Octopus Pro. The other, older version, is correct.

GadgetAngel, 2022-07-05:
The color PIN diagram was based on a PIN.pdf file that was created on 10/17/2021.  The current PIN.pdf file was updated on 12/9/2021 and unfortunately I did not see that change.  BTT does not have any obligation to notify me of there updates and I though I checked for changes but I missed that change.  

Sorry if this caused you any inconvenience.  I will update the color PIN diagram on my Github repo to reflect this new information.  

Thankyou for making me aware of the error.  Give me a day to make the changes to the color pin diagram.  You will be able to see the changes if you go to: https://github.com/GadgetAngel/BTT_Octopus_Color_PIN_Diagram

P.S.  I just noticed that the Octopus V1.0/V1.1 Color PIN diagram is also effected by this so, Thank you again for catching this error.

GadgetAngel, 2022-07-05:
> The PE11/GND pins are reversed in the color version of the pin diagram for Octopus Pro. The other, older version, is correct.

Correction have been made, please see: https://github.com/GadgetAngel/BTT_Octopus_Color_PIN_Diagram#a-picture-of-the-color-pin-diagram-for-the-octopus-pro-v10

intector, 2023-02-21:
Hey,
this is also reversed on the detailed pinout diagram.