bigtreetech/BIGTREETECH-OCTOPUS-Pro #21: Driver 4 (Motor_3) E0 Works like other Z2

JORGENIGMA, 2022-11-24:
I have the Octopus Pro v1.0 board with the stm32F429 to replace the original board of a Sunlu S8 Plus. Pretty much the rest of the sunlu S8 is original, so by configuring the Octopus Pro with 1 x driver, 1 y driver, 2 z drivers, and 1 extruder driver, the extruder is the same as the z-axis ones, like if it worked in parallel, and it seems to me that it specifically copies the z2 driver, since by disabling Z2_DRIVER_TYPE the extruder driver stops working completely, along with the z2 driver.
This issue occurs in firmware 2.0.9.3, 2.0.9.5, 2.1.x, 2.1.1
I would appreciate the help.
These are my configuration files:
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro/files/10081985/Configuration.zip)

DaLiV, 2022-11-27:
so - default motor definiotions in marlin for that board may be written as :
X - Y - Z-Z1 (that splitted electricaly - need to take in account half of current when in use) - Z2 - E0 - E1 - E2 - E3

For making Z2 working - need adapt config,config-adv ans board pins definition file in marlin fw.

JORGENIGMA, 2022-11-28:
Hola!, Recién adquirí una placa de estas para sustituir la original de la SUNLU S8 PLUS, pero me encontré con el problema de que al seguir el tutorial para la Octopus Pro , redirige al tutorial de la Octopus normal (https://3dwork.io/btt-octopus) el Extrusor no funciona, o mejor dicho funciona como un tercer motor para el eje Z (se movía el motor cada que movía el eje Z) revisé la configuración de pines, y tuve que modificar el archivo pins_BTT_OCTOPUS_V1_common.h, porque el motor Z2 es directamente controlado por el mismo controlador de Z1, por lo que para el firmware solo hay un eje Z, asi que por firmware NO HAY QUE DEFINIR Z2!!! y hay que realizar un cambio de nombre a los pines asignados a Z2, ya que repito, para la OCTOPUS PRO se manejan dos motores con el mismo Z_DRIVER, por lo tanto los pines de Z2 los pasé a nombrar de Z2_XXXX_XXX a E0_XXXX_XXX, y el resto de extrusores los desplazados, de modo que el original E0 ahora es E1, el E1 es E2, etc. Con esos cambios la placa funciona genial.
Esta información de los cambios en el archivo de pines no lo conseguí en ningún lado, así que lo publico aquí por si alguien mas se compra esta placa.
Me gustaría también que agregaran estos pasos extra a este tutorial, pero por si no soy leído por algún administrador aquí deje la solución.

Anexo los archivos que modifiqué.

Hello!, I just bought one of these boards to replace the original one for the SUNLU S8 PLUS, but I ran into the problem that when I follow the Octopus Pro tutorial, it re
[Marlin-2.1.1-Mod-BTT-Octopus-Pro.zip](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro/files/10107194/Marlin-2.1.1-Mod-BTT-Octopus-Pro.zip)
directs to the normal Octopus tutorial (https://3dwork. io/btt-octopus) the Extruder does not work, or rather it works as a third motor for the Z axis (the motor moved every time I moved the Z axis) I checked the pin configuration, and I had to modify the file pins_BTT_OCTOPUS_V1_common.h , because the Z2 motor is directly controlled by the same Z1 controller, so for the firmware there is only one Z axis, so by firmware Z2 DOES NOT HAVE TO BE DEFINED!!! and it is necessary to make a name change to the pins assigned to Z2, since I repeat, for the OCTOPUS PRO two motors are managed with the same Z_DRIVER, therefore the Z2 pins were named from Z2_XXXX_XXX to E0_XXXX_XXX, and the the rest of the extruders are displaced, so that the original E0 is now E1, E1 is E2, etc. With those changes the board works great.
I couldn't find this information about the changes in the pin file anywhere, so I'm posting it here in case someone else buys this board.
I would also like them to add these extra steps to this tutorial, but in case I am not read by an administrator, I will leave the solution here.

I am attaching the files that I modified.

closed 2022-11-28 by JORGENIGMA

DaLiV, 2022-11-29:
exactly "Z1" is not defined in any ways ... but Z2 is second Motor for Z axis (with idea that hardwired Z1 can use only HALF of current ... and that may be not enough for some steppers ... )

looking at Marlin/src/pins/stm32f4/pins_BTT_OCTOPUS_V1_common.h /Motor comments/
and count headers - you will get
defaults **X Y (Z Z') Z2 E0 E1 E2 E3** = 9 Motor Headers = 8 Motor Definitions /X Y Z Z2 E0 E1 E2 E3/ in before named file ...
X = // MOTOR 0
Y = // MOTOR 1
Z = // MOTOR 2
Z2 = // MOTOR 3
E0 = // MOTOR 4
E1 = // MOTOR 5
E2 = // MOTOR 6
E3 = // MOTOR 7

they must not be read as
<del>X Y Z Z1 Z2 E0 E1 E2 E3 E4</del>= 10 Motor Headers ... board has still 8 drivers ... not 9 ...
then all will be understandable ... and will not be <del>E0 is at E1</del> ...

also as inconsistancies that someone provide in theri "tutorials" are not related to Marlin repository ...
simple read a code + Take a look into boards Draw-Schematics