bigtreetech/BIGTREETECH-OCTOPUS-Pro #23: no spi communication for drivers - missing PA6

SaPu84, 2022-12-12:
I get a new H723 and find a broken spi1 connection on the PCB to the driver.
Workaround is to use PA6 of EXP2 and connect it to the driver line by using a jumper cable.
May it run into other electrical issues by doing it?

MAX31865 is working fine.

BR
Sascha

jmeier5261, 2023-01-08:
Noted the same on my h723 v1.0.1 octopus pro. Drivers do not appear to connect to spi bus pa6 pin. MAX m31865 works fine but there appears to be an error on the pcb itself and the driver sockets are not wired to pa6. If I run a jumper from PA6 to the driver pin it works as expected. @bigtreetech was this fixed in a later hardware revision or just a bad batch of boards?

kmobs, 2023-02-21:
Any official word on this?

jmeier5261, 2023-02-21:
Nothing official just return the board to the vendor for a replacement if possible. There’s a bunch of these dud boards out there.

kmobs, 2023-02-21:
Thanks. I ended up just tapping into the LCD pin like y'all said. This + the fact that it enables the heater pin in DFU makes this board extremely disappointing. I won't be recommending it to anyone.

bigtreetech, 2023-02-23:
The diodes used previously were not very high in parameters and there was a chance that they would break through when the customer was using them in a harsh environment (high static, humidity, etc.), resulting in a failure to communicate with the driver SPI signal. We have now addressed this problem.

kmobs, 2023-02-24:
I'm sorry but I don't know if this is the case? I never had communication on the bus.

How have you resolved the issue?

NAPCAL, 2023-03-22:
I have checked both of my Pro boards and found no continuity issues with the SPI1_MISO or Drivers MISO. Here is what I checked on my F446 & H723.

[image]

[image]

csdstudio78, 2023-03-23:
> 

Same... brand new out of the box Octo Pro with 446, no SPI1.

NAPCAL, 2023-03-23:
@csdstudio78 Can you verify that your U27 and D41 are facing the same way as in my photo?

csdstudio78, 2023-03-23:
> @csdstudio78 Can you verify that your U27 and D41 are facing the same way as in my photo?

[image]

csdstudio78, 2023-03-23:
To add, connecting stepper drivers to PA6 via EXP2 enables spi1.

NAPCAL, 2023-03-23:
@csdstudio78 Thanks for the photo; it seems to be a revision of the PCB that the trace between D21 and pin 6 of U27 is incorrect.

My good board has the trace, and the bad PCB has D21 going to a via, but it doesn't seem to connect to U27 pin 6.

The PCB manufacturer may have made this error on the current batch of PCBs.

Good board
[image]

Bad board
[image]

csdstudio78, 2023-03-23:
Ahh yes. OK. Thanks for confirming! I definitely have a bad board... I've already requested a replacement via Amazon so we'll see if that one is any different.  :)

NAPCAL, 2023-03-24:
I forgot to post the board version of my good boards. @csdstudio78 Could you check and post your board version?

My good Octopus Pro’s
[image]
[image]

csdstudio78, 2023-03-24:
OK... I hate to burst our bubbles, but I went back today and looked at my board again, knowing what you were looking for, and with the camera flash there's definitely a trace.  Time to do some more checking.

[image]

NAPCAL, 2023-03-24:
@csdstudio78 Any chance you can ohm out the paths?

The jump wire from EXP2 MISO to the drivers leaves the STM unprotected by D21.

If you can find the failure location I can check mine.

csdstudio78, 2023-03-24:
> @csdstudio78 Any chance you can ohm out the paths?
> 
> The jump wire from EXP2 MISO to the drivers leaves the STM unprotected by D21.
> 
> If you can find the failure location I can check mine.

Agreed. I'll pull my board today and look. I don't like using exp2 either.

csdstudio78, 2023-03-24:
> @csdstudio78 Any chance you can ohm out the paths?
> 
> The jump wire from EXP2 MISO to the drivers leaves the STM unprotected by D21.
> 
> If you can find the failure location I can check mine.

OK. All traces check out. Power checked at 3.3v, 5.0v, and GND, all points verified in the schematic you posted above. Diode checks out, all resistor values checked. Because I'm not an EE, I can't figure out why this doesn't work other than not understanding U27. (its role and whether its oriented correctly)

csdstudio78, 2023-03-24:
@NAPCAL 

I wonder if U27 is backwards, shouldn't pin 6 be the output feeding MISO?
[image]

csdstudio78, 2023-03-24:
@NAPCAL  See below... am I wrong here?

[image]

NAPCAL, 2023-03-24:
@csdstudio78, the circuit is a level shifter for Master In Slave Out from the drivers @ 5V; R203 ensures the input to the buffer never floats; R205 is a 3.3V pull-up to also ensures that the STM input never floats; U27 is only to translate low signals through to the STM SPI1_MISO pin 42 by the use of the D21 in a reverse direction (DIODE SCHOTTKY 30V 200MA).

[image]

NAPCAL, 2023-03-24:
@csdstudio78 Also, note the other SPI1 signals go through U26 of the same device, and it is connected to +5V, so the +3.3V signal from the STM will get stepped up to +5V. It is when you are stepping down that the extra components are required.

csdstudio78, 2023-03-24:
@NAPCAL ... Ok, I'm tracking now.  So the schematic is good and the physical layout appears to align with the schematic.  We're running out of options.

csdstudio78, 2023-03-24:
Forward bias of my diode (D41) is 0.27v, resistance is 92k.  Negative probe on the left side, towards U27.

NAPCAL, 2023-03-24:
@csdstudio78 D21
Readings taken with a Fluke 117

Board version 1.0 bias voltage 0.238V 65.6K ohm
Board version 1.0.1 bias voltage 1.685V 3.17Meg ohm

I haven't tried my Version 1.0.1 with TMC5160 Pro's yet.

csdstudio78, 2023-03-24:
@NAPCAL  Your rev 1.0 readings are similar to my rev 1.0.1 readings above for the diode. Just to verify, you're checking D41, not D21, correct?  BTT TMC5160 Pros work ok with MISO jumped to PA6 on EXP2.  I initially dropped in a set of new Fysetc QHV5160's and both sense resistors on one driver blew within a few minutes, current limit set to 1.5A, motor voltage at 50v...lol.

NAPCAL, 2023-03-24:
Yes D41, type-o
Just tested both of my Pro boards with TMC5160 Pro (X & Y) and TMC2209 (Z & Extruder)
All four responded 

15:43:50 $ DUMP_TMC STEPPER=stepper_x
15:43:50 // ========== Write-only registers ==========
15:43:50 // GLOBALSCALER: 00000043 globalscaler=67
15:43:50 // IHOLD_IRUN: 00061f13 ihold=19 irun=31 iholddelay=6
15:43:50 // MSLUT0: aaaab554 mslut0=2863314260
15:43:50 // MSLUT1: 4a9554aa mslut1=1251300522
15:43:50 // MSLUT2: 24492929 mslut2=608774441
15:43:50 // MSLUT3: 10104222 mslut3=269500962
15:43:50 // MSLUT4: fbffffff mslut4=4227858431
15:43:50 // MSLUT5: b5bb777d mslut5=3048961917
15:43:50 // MSLUT6: 49295556 mslut6=1227445590
15:43:50 // MSLUT7: 00404222 mslut7=4211234
15:43:50 // MSLUTSEL: ffff8056 w0=2 w1=1 w2=1 w3=1 x1=128 x2=255 x3=255
15:43:50 // MSLUTSTART: 00f70000 start_sin90=247
15:43:50 // TPWMTHRS: 000fffff tpwmthrs=1048575
15:43:50 // COOLCONF: 00000000
15:43:50 // PWMCONF: c40c001e pwm_ofs=30 pwm_autoscale=1 pwm_autograd=1 pwm_reg=4 pwm_lim=12
15:43:50 // TPOWERDOWN: 0000000a tpowerdown=10
15:43:50 // ========== Queried registers ==========
15:11:31 // MSLUT7: 00404222 mslut7=4211234
15:43:50 // GCONF: ffffffff recalibrate=1 faststandstill=1 en_pwm_mode=1 multistep_filt=1 shaft=1(Reverse) diag0_error=1 diag0_otpw=1 diag0_stall=1 diag1_stall=1 diag1_index=1 diag1_onstate=1 diag1_steps_skipped=1 diag0_int_pushpull=1 diag1_poscomp_pushpull=1 small_hysteresis=1 stop_enable=1 direct_mode=1 test_mode=1
15:11:31 // MSLUTSEL: ffff8056 w0=2 w1=1 w2=1 w3=1 x1=128 x2=255 x3=255
15:43:50 // CHOPCONF: ffffffff toff=15 hstrt=7 hend=15 fd3=1 disfdcc=1 chm=1 tbl=3 vhighfs=1 vhighchm=1 tpfd=15 mres=15(0usteps) intpol=1 dedge=1 diss2g=1 diss2vs=1
15:11:31 // MSLUTSTART: 00f70000 start_sin90=247
15:43:50 // GSTAT: ffffffff reset=1(Reset) drv_err=1(ErrorShutdown!) uv_cp=1(Undervoltage!)
15:11:31 // TPWMTHRS: 000fffff tpwmthrs=1048575
15:43:50 // DRV_STATUS: ffffffff sg_result=1023 s2vsa=1(ShortToSupply_A!) s2vsb=1(ShortToSupply_B!) stealth=1 fsactive=1 csactual=255 stallguard=1 ot=1(OvertempError!) otpw=1(OvertempWarning!) s2ga=1(ShortToGND_A!) s2gb=1(ShortToGND_B!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) stst=1
15:11:31 // COOLCONF: 00000000
15:43:50 // FACTORY_CONF: ffffffff factory_conf=31
15:11:31 // PWMCONF: c40c001e pwm_ofs=30 pwm_autoscale=1 pwm_autograd=1 pwm_reg=4 pwm_lim=12
15:43:50 // IOIN: ffffffff refl_step=1 refr_dir=1 encb_dcen_cfg4=1 enca_dcin_cfg5=1 drv_enn=1 enc_n_dco_cfg6=1 sd_mode=1 swcomp_in=1 version=0xff
15:11:31 // TPOWERDOWN: 0000000a tpowerdown=10
15:43:50 // LOST_STEPS: ffffffff lost_steps=1048575
15:11:31 // ========== Queried registers ==========
15:11:31 // ========== Queried registers ==========
15:43:50 // MSCNT: ffffffff mscnt=1023
15:11:31 // GCONF: 0000000c en_pwm_mode=1 multistep_filt=1
15:11:31 // GCONF: 0000000c en_pwm_mode=1 multistep_filt=1
15:43:50 // MSCURACT: ffffffff cur_a=-1 cur_b=-1
15:11:31 // CHOPCONF: 34410153 toff=3 hstrt=5 hend=2 tbl=2 tpfd=4 mres=4(16usteps) intpol=1 dedge=1
15:11:31 // CHOPCONF: 34410153 toff=3 hstrt=5 hend=2 tbl=2 tpfd=4 mres=4(16usteps) intpol=1 dedge=1
15:43:50 // OTP_READ: ffffffff otp_fclktrim=31 otp_s2_level=1 otp_bbm=1 otp_tbl=1
15:11:31 // GSTAT: 00000005 reset=1(Reset) uv_cp=1(Undervoltage!)
15:11:31 // GSTAT: 00000005 reset=1(Reset) uv_cp=1(Undervoltage!)
15:43:50 // PWM_SCALE: ffffffff pwm_scale_sum=255 pwm_scale_auto=-1
15:11:31 // DRV_STATUS: 80134000 stealth=1 csactual=19 stst=1
15:11:31 // DRV_STATUS: 80134000 stealth=1 csactual=19 stst=1
15:43:50 // PWM_AUTO: ffffffff pwm_ofs_auto=255 pwm_grad_auto=255
15:11:31 // FACTORY_CONF: 0000000f factory_conf=15
15:11:31 // FACTORY_CONF: 0000000f factory_conf=15
15:43:50 // TSTEP: ffffffff tstep=1048575
15:11:31 // IOIN: 30000050 drv_enn=1 sd_mode=1 version=0x30
15:11:31 // IOIN: 30000050 drv_enn=1 sd_mode=1 version=0x30
15:11:31 // LOST_STEPS: 00000000
15:11:31 // LOST_STEPS: 00000000
15:11:31 // MSCNT: 00000008 mscnt=8
15:11:31 // MSCNT: 00000008 mscnt=8
15:11:31 // MSCURACT: 00f7000c cur_a=12 cur_b=247
15:11:31 // MSCURACT: 00f7000c cur_a=12 cur_b=247
15:11:31 // OTP_READ: 0000000f otp_fclktrim=15
15:11:31 // OTP_READ: 0000000f otp_fclktrim=15
15:11:31 // PWM_SCALE: 00000012 pwm_scale_sum=18
15:11:31 // PWM_SCALE: 00000012 pwm_scale_sum=18
15:11:31 // PWM_AUTO: 0000001d pwm_ofs_auto=29
15:11:31 // PWM_AUTO: 0000001d pwm_ofs_auto=29
15:11:31 // TSTEP: 000fffff tstep=1048575
15:11:31 // TSTEP: 000fffff tstep=1048575

15:46:19 $ DUMP_TMC STEPPER=stepper_z
15:46:19 // ========== Write-only registers ==========
15:46:19 // SLAVECONF: 00000200 senddelay=2
15:46:19 // IHOLD_IRUN: 00081414 ihold=20 irun=20 iholddelay=8
15:46:19 // TPWMTHRS: 00000000
15:46:19 // TPOWERDOWN: 00000014 tpowerdown=20
15:46:19 // SGTHRS: 00000000
15:46:19 // ========== Queried registers ==========
15:46:19 // GCONF: 000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
15:46:19 // GSTAT: 00000001 reset=1(Reset)
15:46:19 // IFCNT: 00000008 ifcnt=8
15:46:19 // OTP_READ: 0000000f otp_fclktrim=15
15:46:19 // IOIN: 21000041 enn=1 pdn_uart=1 version=0x21
15:46:19 // FACTORY_CONF: 0000000f fclktrim=15
15:46:19 // TSTEP: 000fffff tstep=1048575
15:46:19 // MSCNT: 00000028 mscnt=40
15:46:19 // MSCURACT: 00ef003c cur_a=60 cur_b=239
15:46:19 // CHOPCONF: 34030053 toff=3 hstrt=5 tbl=2 vsense=1 mres=4(16usteps) intpol=1 dedge=1
15:46:20 // DRV_STATUS: c0140000 cs_actual=20 stealth=1 stst=1
15:46:20 // PWMCONF: c80d0e24 pwm_ofs=36 pwm_grad=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 pwm_reg=8 pwm_lim=12
15:11:31 // MSLUT7: 00404222 mslut7=4211234
15:46:20 // PWM_SCALE: 00000017 pwm_scale_sum=23
15:11:31 // MSLUTSEL: ffff8056 w0=2 w1=1 w2=1 w3=1 x1=128 x2=255 x3=255
15:46:20 // PWM_AUTO: 000e0024 pwm_ofs_auto=36 pwm_grad_auto=14
15:11:31 // MSLUTSTART: 00f70000 start_sin90=247
15:46:20 // SG_RESULT: 00000000

csdstudio78, 2023-03-24:
Alright! I'm getting somewhere.  Board rev 1.0.1 SPI1 works as long as there's no display connected to EXP2 (Small bummer).  There's some sharing issues with SPI1 on the Octo Pro that aren't present on the non-pro Octo.  That Octo board doesn't have the level shifting logic, it's straight from stepper driver to PA6.

NAPCAL, 2023-03-24:
@csdstudio78 
Well, the wire jump would cause a pull-up to +5V due to the drivers at +5V logic (MISO) and the EXP2 (SPI1_MISO) at +3.3V; without the wire jump, the display seams to pull the logic levels into a between state even though doesn't use (SPI1).

jessepietila, 2023-05-04:
Hi,

I have Pro1.0.1 F429 board, not able to run E with TMC5160 Pro v1.1.....

So what to do ?

tutankino, 2023-05-26:
Hi all, i have the same problem, board h723 1.01 and also with the jumper on exp2 pa6 i received Always SPI global scaler error, any suggestion?

rexsham, 2023-06-14:
> Hi all, i have the same problem, board h723 1.01 and also with the jumper on exp2 pa6 i received Always SPI global scaler error, any suggestion? Problem Is d41??? I changed d41 with 1n4148 same as a new revision 1.1, no success, not work..😭

An even worse situation is, I am using Revision 1.1, and tmc5160pro, still don't work

looxonline, 2023-08-07:
> An even worse situation is, I am using Revision 1.1, and tmc5160pro, still don't work

Did you figure out what was happening with your board? Do you have photos of the version number on your board?

ansonl, 2023-08-07:
I have board v1.0.1 and my workaround is in #31

looxonline, 2023-08-07:
> I have board v1.0.1 with TMC2209 and the MAX pluggable stepper module and my workaround is in #31

Yours seems to be a uniquely Marlin issue. I think that, while similar on the surface, these are not the same issues.

NAPCAL, 2023-08-07:
@ansonl 
> I have board v1.0.1 with TMC2209 and the MAX pluggable stepper module and my workaround is in #31 

TMC2209 drivers don't use SPI they are only UART.

ansonl, 2023-08-07:
@NAPCAL the BTT MAX31865 plug in stepper driver module that I am using communicates over SPI.

NAPCAL, 2023-08-07:
@ansonl 

Thanks for adding that, are you using the jumper from EXP2 to one of the Driver sockets?

[image]

ansonl, 2023-08-07:
@NAPCAL I don't have it jumped like you specified. I have the MAX module inserted like normal into an unused stepper slot and I have an LCD SD reader (stock ultimaker 2 display defined as ULTI_CONTROLLER) communicating through EXP2.

NAPCAL, 2023-08-07:
@ansonl 

Thanks
What firmware are you using?

ansonl, 2023-08-07:
I am using Marlin bugfix-2.1.x from around this June. 

https://github.com/ansonl/Marlin/tree/dxu-ultimaker-btt-octopus-pro-1

exsencer, 2023-08-13:
> @ansonl
> 
> Thanks for adding that, are you using the jumper from EXP2 to one of the Driver sockets?
> 
> [image]

Is this the final solution? I have a 1.1 board I am trying to get up and running with 5160's and I don't think they fixed this issue with this revision. The pins have been swapped around though so that the heater is not on during DFU mode

NAPCAL, 2023-08-13:
That is a temporary fix, but we need to get everyone that has to do this to get their TMC5160 or other SPI TMCs, to contact BTT support so they can see it is a problem for board owners.

pasha4ur, 2024-02-29:
Hello

BTT Octopus Pro F429 1.0.1 + BTT TMC 2240 (SPI) + BTT 12864 mini v2

After 3 print jobs:

Unable to write tmc spi ‘stepper_x’ register GLOBALSCALER
Unable to write tmc spi ‘stepper_x’ register GLOBALSCALER
Unable to write tmc spi ‘stepper_x’ register GLOBALSCALER

Unable to write tmc spi ‘stepper_x’ register GCONF

Only disconnecting BTT 12864 mini v2 (I tried several) helps.

looxonline, 2024-02-29:
> Hello
>
> BTT Octopus Pro F429 1.0.1 + BTT TMC 2240 (SPI) + BTT 12864 mini v2
>
> After 3 print jobs:
>
> Unable to write tmc spi ‘stepper_x’ register GLOBALSCALER Unable to write tmc spi ‘stepper_x’ [...]

Do you mean that it works after you disconnect the display? BTW, Please contact the official support email as the github issue list is not used for official support tickets.

pasha4ur, 2024-02-29:
> Do you mean that it works after you disconnect the display? BTW, Please contact the official support email as the github issue list is not used for official support tickets.

If display connected to EXP I can't do anything with motors. Even homing.

If I disconnected display it makes homing wo errors.

I printed 3 times with display. But after last turning on the printer it doesn't work with display anymore. I tried several new BTT 12864 mini (v1 and v2).

As I saw from other forums Bigtreetech provide bad customer support (example on their forum  https://community.biqu3d.com/topic/665-bigtreetech-tmc5160-pro-v11-and%C2%A0bigtreetech-btt-octopus-pro-f446-on-klipper-spi-not-working/). I wrote them with 1 year ago about another product - 0 answers.

I will open ticket. But I have no hope for quality support from a Chinese company.

pasha4ur, 2024-03-01:
Hello

 I have found the source of my printer problem to be probably the location of the screen cable.

https://klipper.discourse.group/t/spi-cant-write-to-tmc2240-after-few-print-jobs/14531/10?u=pasha4ur

pasha4ur, 2024-03-04:
> Do you mean that it works after you disconnect the display? BTW, Please contact the official support email as the github issue list is not used for official support tickets.

I opened ticket with Bigtreetech. They didn't even answer. This is their level of customer support. :)

looxonline, 2024-03-05:
> I opened ticket with Bigtreetech. They didn't even answer. This is their level of customer support. :)

Because we saw your comment on facebook that it was actually your own installation fault (using a cable that is too long) and we closed the ticket. We do not have the resources to help debug issues caused by unique implementations of our products that users are attempting. Each support person costs $15-$20 an hour. That is more profit than we make on a board. Our support is therefore primarily for warranty issues. If users would like implementation support (as is your situation) then we will be launching a paid support tier in the future.

pasha4ur, 2024-03-05:
> Because we saw your comment on facebook that it was actually your own installation fault (using a cable that is too long) and we closed the ticket. We do not have the resources to help debug issues [...]

Great. Closed without notice.

It's not my fault. Your description does not indicate a limit on cable length. The problem can also arise from the intersection of the shield cable with the motor or power cables.

Here people have a standard screen cable, but they also have problems.

Microcontrollers have 4 or 6 SPI buses, but you attached the screen and drivers to 1 spi bus.

csdstudio78, 2024-03-06:
Wtf Lol. They won't offer support because the hourly wage to support you is more than the profit for that one board? That's the most ass-backwards thing I've ever heard. Chinese companies, still living up to their stereotype.

looxonline, 2024-03-06:
> Wtf Lol. They won't offer support because the hourly wage to support you is more than the profit for that one board? That's the most ass-backwards thing I've ever heard. Chinese companies, still living up to their stereotype.

Not sure if you didn't read my response or just chose to misread it. Either way, you are welcome to an opinion but not welcome to drop racist remarks in the comments. Mutual, respectful communication is everyone's right.

Let me rephrase my comment to make it clearer. We offer *warranty* support on our products. This means that if there is a factory fault we will support the user to diagnose it and then offer a replacement or a suitable solution.

We do not officially offer implementation support (although sometimes our agents can help if they are not overloaded). Why? Implementation support means offering a user support in *implementing* the product that they have purchased within their unique use case. Let's use this as an example. Here, a user was trying to use an SPI cable that was not supplied with the product that he purchased. He replaced it with one that he purchased which is too long for an SPI bus and the screen malfunctioned. Correct shielding of the cable and isolation from noise sources was the solution which the user was able to arrive at. The hardware was not at fault. Since DIY printer building is something that involves a lot of hands on tinkering, there are often cases where users try something and then it does not work. A natural reaction is to reach out to the hardware manufacturer and try to get a solution to this but (and here is my point) the economics simply do not allow for us as a hardware provider to offer this kind of *implementation* support. 

There are hardware providers who offer pretty much the same components on the PCB but the board is priced 4x the price and they offer implementation support if it is needed.

Nevertheless, we do have several initiatives planned to offer implementation support to users in a way that will still work with the economics:

1.) A wiki - This has been in development for over a year and is now available at https://bigtreetech.github.io/docs/
2.) Existing manuals and documentation - There is a vast library of existing manuals and documentation
3.) Youtube content with video guides and tutorials - We have invested in a new studio and are busy creating guides and tutorials to help users with their implementations.
4.) A paid implementation support package - Support hours could be purchased and then a dedicated support agent would be assigned to you to help you implement the product. If it is found that one of our products is actually at fault and it is a warranty support issue then the full amount would be refunded. This is still in the works.

This is an entirely fair way of offering a balance between support and product cost.

closed 2024-03-06 by bigtreetech

pasha4ur, 2024-03-07:
You even added racism to this topic. :D

People have problems with your boards not because of their errors, but because you hung the screen, drivers, etc. on only 1 spi bus.

csdstudio78, 2024-03-07:
> You even added racism to this topic. :D
> 
> People have problems with your boards not because of their errors, but because you hung the screen, drivers, etc. on only 1 spi bus.

This.

looxonline, 2024-03-07:
> You even added racism to this topic. :D
> 
> People have problems with your boards not because of their errors, but because you hung the screen, drivers, etc. on only 1 spi bus.

SPI is designed for multi slave on a single bus. By you changing the cable to a longer one and not following proper routing techniques, you introduced too much noise to the bus. Even if the drivers were not on the same bus your screen would not have worked correctly.

Nevertheless, the schematics are open source so you are welcome to review them and see for yourself that every single pin on the MCU is used with many of the hardware SPI busses used for other purposes. Multiplexing is both a part of the spec, and required in many cases.

pasha4ur, 2024-03-07:
> SPI is designed for multi slave on a single bus. By you changing the cable to a longer one and not following proper routing techniques, you introduced too much noise to the bus. Even if the drivers [...]

People in this topic have problems with spi even with standard display cable. There are a lot of such topics in the Internet from people with standard cable

Will try boards from other vendor next time.

jmacheta, 2024-03-20:
**Please do not directly connect the stepper motor MISO pin to the SPI_MISO pin on the EXT2 header.**

The reason is that the TMC outputs 5V, while the STM32F4 uses 3.3V logic., and it may tolerate at most 4V at any given pin.
This  poses a risk of damaging your CPU and/or any other 3.3V devices on the SPI bus (such as an LCD display).

Instead, consider replacing D41 with a 100-ohm or 1k-ohm resistor (0603 size fits well). This will create a direct path between the gate’s output and the CPU input, limiting the eventual current (which should be negligible). I’ve personally used this approach, and it works well.

Alternatively, you can place a discrete level shifter between the stepper motor’s MISO and the EXT2 SPI_MISO pin. While this solution works, I find dangling cables less appealing.

ansonl, 2024-05-04:
@looxonline What is the maximum recommended cable length for the EXP1 and EXP2 cables? I had to make a converter board to remap the pins for I2C to use with the Ultimaker Ulticontroller display controller and sometimes the built in SD card has an error that crashes Marlin as previously described in my post for this thread.