bigtreetech/BIGTREETECH-OCTOPUS-Pro #25: The SPI bus cannot communicate with the “TMC5160 PRO” driver.

mekaireb, 2023-02-24:
Issue:
The SPI bus cannot communicate with the “TMC5160 PRO” driver.

Notes:
• STEP/DIR work.
• Not able to communicate with SPI
• Tested two TMC5160 PRO drivers with the same result.
• Octopus Pro V1.0 STM32F429ZGT6

Log:
Connecting...
//action:prompt_end
Testing Z connection... Error: All HIGH
//action:notification TMC CONNECTION ERROR
echo:SD card ok
ok B:-49.10 /0.00 @:0 B@:0
Printer is now online.
ok B:-49.10 /0.00 @:0 B@:0
>>>M122 
SENDING:M122
                        Z
Enabled                       false
Set current      800
RMS current    790
MAX current    1114
Run current     23/31
Hold current    11/31
Global scaler   130/256
CS actual        31/31
PWM scale      4294967295
vsense
stealthChop     true
msteps            0
interp               true
tstep                4294967295
PWM thresh.   0
[mm/s]             -
OT prewarn     true
triggered
OTP                false
off time            15
blank time       54
hysteresis
-end                12
-start               8
Stallguard thrs 0
uStep count     65535
DRVSTATUS  Z
sg_result         1023
stallguard        *
fsactive            *
stst
olb                   *
ola                   *
s2gb                *
s2ga                *
otpw                *
ot                     *
Driver registers:
                        Z          0xFF:FF:FF:FF            Bad response!
Testing Z connection... Error: All HIGH
//action:notification TMC CONNECTION ERROR

[image]

[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro/files/10820137/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro/files/10820138/Configuration_adv.h.txt)

mjfsch, 2023-02-24:
are your drivers configured for UART or SPI? the TMC5160 can support both, but your specific drivers will be setup for one or the other, possibly neither

sometimes you are required to solder a small link on the driver PCB to select the mode, might be worth checking the datasheet for the drivers

mmakam2, 2023-02-25:
I had the same issue, the board is missing SPI traces. You have to jump a wire from the EXP2 MOSI pin to a spare stepper slot pin SDO. Make sure the spare slot is in SPI mode. PA6 is the top left pin on the EXP2 connector.

NAPCAL, 2023-03-22:
I have checked both of my Pro boards and found no continuity issues with the SPI1_MISO or Drivers MISO. Here is what I checked on my F446 & H723. 
[image]
[image]

NAPCAL, 2023-03-24:
I may have found the issue with the PCB; in a photo provided by @csdstudio78, there seems to be a misrouted trace between D21 and pin 6 of U27.

Board having issues with SPI1_MISO
[image]

My boards with no issues.
[image]
[image]
[image]

jessepietila, 2023-05-04:
> I had the same issue, the board is missing SPI traces. You have to jump a wire from the EXP2 MOSI pin to a spare stepper slot pin SDO. Make sure the spare slot is in SPI mode. PA6 is the top left pin on the EXP2 connector.

-------- I think I would need to do this. ...Or otherwise I go for SD mode on Marlin ?

tutankino, 2023-05-26:
Any news? I have the same problem on octopus pro h723 1.0.1 and jumper tò exp2 pa6 not work

jessepietila, 2023-05-26:
I have Octopus Pro V1.0.1 F429 and 5160 ProV1.1 (for extruder),

I have not still solved this issua / situation.

At the moment I am using Marlin 2.1.2

Looking forward to hear more from You,

Jesse