bigtreetech/BIGTREETECH-OCTOPUS-Pro #26: 无法写入tmc spi“stepper_x”寄存器GLOBALSCALER

asdfsgasdfa, 2023-04-01:
BTT 章鱼pro1.0.1版本
FLy外置5160pro
[image]
[image]

asdfsgasdfa, 2023-04-01:
我实在不知道为什么会这样

asdfsgasdfa, 2023-04-01:
我实在不知道为什么会这样
[image]

asdfsgasdfa, 2023-04-01:
如第二张图所示，我甚至使用了旧版本（1.0版本）的spi错误解决方法，驱动的SDO飞线到EXP2的PA6引脚并重启，但是无济于事

tutankino, 2023-05-26:
Any news?  I have same problema on octopus pro h723 1.0.1

asdfsgasdfa, 2023-05-26:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

rexsham, 2023-06-14:
> Any news? I have same problem on octopus pro h723 1.0.1

Same issue here