bigtreetech/BIGTREETECH-OCTOPUS-Pro #27: USB-C power and serial for Raspberry Pi 4

Kaelum, 2023-04-05:
With the **MCU Power Jumper** in place can the USB-C connector be used as both power to a Raspberry Pi 4 and for serial communications.  It looks like it can, but I don't see this mentioned anywhere.

Kaelum, 2023-04-06:
I see that you placed D35 inline with the 5V supply to prevent current from flowing out of the USB-C connector.  So, there are 2 options:

1. Short D35.
2. Place a jumper on the USB-C side of J68 to +5v.

I tried the latter, running a jumper from the _5v on one of the fan, and it works.  It would be nice if a future revision had a jumper to short D35, which would allow people to power and connect to their Raspberry Pi directly from the USB-C port of the Octopus Pro.

miklschmidt, 2023-04-10:
@Kaelum i don't think you're supposed to do that. There are issues when power is flowing back through the USB-A ports on the pi (for example `reboot` doesn't work), and the USB-C port on the pi is power only.

ayufan, 2023-04-26:
@miklschmidt The RPI 4B USB-C is in fact USB OTG port.

miklschmidt, 2023-04-26:
@ayufan ah, so it is!

Kaelum, 2023-04-27:
Yeah, in the end I used 2 USB-C splitter/combiner boards and a dedicated 5v power supply, which works perfectly.  I basically have a remote RPi Tablet connected to my Rat Rig, which also uses a CAN tool head.  I needed the separate power supply because of the voltage drop over the USB-C cable, which is very significant when powering both the RPi and Official Screen.

I tried running RatOS at first, but that was a complete waste of time.  RatOS does not work with CAN, and it has several other very serious design flaws which actually destroyed some of the printed parts.  I ended up using MailsailOS, and everything just worked so smoothly.  KlipperScreen is freakin' AWESOME with this setup.

closed 2023-04-27 by Kaelum

miklschmidt, 2023-04-27:
@Kaelum It works just fine with CAN like any other installation, multiple users do use CAN with it, i just **heavily** discourage it, because it makes no sense over USB for the supported printers in RatOS (and i'd argue in general).

> and it has several other very serious design flaws which actually destroyed some of the printed parts. 
20k downloads, and only you have these issues. The issues you raised clearly indicated you didn't read the instructions. Use it for what it was meant for, don't try to use a hammer to screw in a screw. Also this is far off topic, i don't know why you feel you needed to rant and publicly flame me for my work here. If you don't like it, fine, but don't air your dirty laundry all around the internet, just because you refuse to understand why things are the way they are. Everything isn't about you and your specific niche use case.

Kaelum, 2023-04-27:
@miklschmidt your comment above is making false assumptions.  I was using RatOS w/o CAN in the beginning, and I followed the instructions verbatim.  After working through several issues I tried moving to CAN.  RatOS does not **FULLY** work with CAN under any circumstance.  As you said, this isn't the place, so it is what it is, and I'm happy with MainsailOS.