bigtreetech/BIGTREETECH-OCTOPUS-Pro #32: Octopus Pro 1.1 (STM32H7) and TMC5160 Pro V1.1  Cannot Communicate

rexsham, 2023-06-14:
I have a bunch of BTT TMC5160 Pro v1.1
I used them on MKS Monster 8v 1.0 without issues.

But on BTT Octopus Pro v1.1  STM32H723 
I got "Unable to write tmc spi 'stepper_z' register" issues
In SPI bus settings I tried software spi pa6 pa5 pa7 and spibus : spi1. I also tried to set spibus speed to various speeds, all yield the same error.

I read that v1.0 and v1.01 versions have the PA6 MISO diode issue. 

I checked the forward bias of my diode is 0.57V, the direction is also correct (cathode properly connecting to buffer chip output and anode to pin 42 of mcu).

Also the motor driver MISO is properly connecting to thr input pin of the buffer chip. 

Can anyone point me towards some directions?
Anyone got success with Octopus Pro v1.1 and TMC5160 Pro v1.1?

2kclay, 2023-08-22:
Did you get a resolution with this issue? I have same issue on Octopus v1.1 non pro but using mellow external 48v input drivers

motymen, 2023-11-23:
are you running marlin on the Octopus Pro 1.1 (STM32H723)?