bigtreetech/BIGTREETECH-OCTOPUS-Pro #36: Marlin Octopus Pro v1.1 (H723ZE) no definition

vfbank, 2023-07-03:
Even in the latest version of the Marlin firmware, there is no definition to support the H723ZE chipset in the boards.h. How do I use it?

Ari-SSO, 2023-07-11:
I dont remember but you must select the BTT_OCTOPUS_PRO_V1_1 or 1_0 as board, and in platformio chispset config you must select "[STM32H723Vx_btt]"

vfbank, 2023-07-12:
> 

I asked the Merlin community and got an answer already.

The V1.1 version has not been merged yet, so we have to wait a little longer.

V1.1 will be available in the next merged version.

Ari-SSO, 2023-07-12:
i not sure, but i read something for a workarround, something like i say before to set the V1_0 and punto STM32H723. Use the night build of marling, the bugfix version

Ari-SSO, 2023-07-13:
In platformio.ini set:
default_envs = STM32H723Zx_btt

Ari-SSO, 2023-07-13:
Or use directly  "STM32H723ZE_btt", this is defined in stm32h7.ini file

destruct123, 2024-04-09:
OCTOPUS pro EZ uses the same STM32H7, declare board as octopus pro EZ, copy the common pins for the octopus pro on stmH7  folder, copy and paste the text for octopus pro V1_1 pins on Octopus pro ez and see the magic happens, just correct the pins as instructed on the github page.