bigtreetech/BIGTREETECH-OCTOPUS-Pro #37: Octopus Pro H729 DIAG pins

PerLangkjaer, 2023-07-08:
I have a problem where the jumpers for the DIAG pins to run sensorless homing, have no connection to the pin of the endstop.

Is the H729 a newer version???

Ari-SSO, 2023-07-11:
J729 or H723?

PerLangkjaer, 2023-07-11:
I have the H729 version.
I have found out that the HE0 and HE2 is not working, but I suspect it have something with the pin names in the config file.

Ari-SSO, 2023-07-11:
you search in bigtreetech github https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0?
or 
https://www.youtube.com/watch?v=v9ahI2IsPV0

Ari-SSO, 2023-07-11:
I read that v1.0 has some problem with pin, they are wrong informed in the board. Refer to the diagrams in PDF. I don't remember but ther is somethig with the - and + pin. This in version 1.1 of octopus was corrected.