bigtreetech/BIGTREETECH-OCTOPUS-Pro #40: BTT Tmc 5160 1.2 and octopus pro giving error "TMC 'stepper_y1' reports error: GSTAT: 00000004 uv_cp=1(Undervoltage!")

Shamoon78, 2023-08-06:
They run for couple second but then this error massage ,and same error after restart.
My thorough research seams 8 mhz crystal on 429 version causing the problem need 12 mhz version please correct me if Iam wrong and help me find a solution.(it is not wires problem wires are all brand new Ldo , and same board and motors run perfectly fine on BTTtmc 2209 with no problems.
Thanks.

PZrk73, 2023-08-07:
I have a similar configuration. Everything works fine. Previously, there were errors on the Z axis, too, with two motors, but with TMC2130 drivers. Sometimes there was a message about a voltage drop, sometimes about the breakage of the motor windings, sometimes about the short-circuit of the windings. Sometimes on one, and sometimes on the second motor. Fixed by setting the option stealthchop_threshold = 0 (disable).

Shamoon78, 2023-08-07:
Hi [PZrk73], it is 0 on all drivers.but the problem still there

feaver12, 2024-08-16:
Same problem here, tmc5160 v1.2 and skr1.4t board. 2209s work fine. Have tried the 5160s in 3 different printers and yeild the same results. Gstat undervoltage error. The axis that throws the error is random. The drivers work fine on marlin. Have had the drivers for years now and have never been able to get them working with klipper. Super frustrating as I have around 20 of them sitting there. Did you manage to get any further with them?

WillGIam7, 2025-04-11:
Did anyone solve that problem? I´m having the same.

bigtreetech, 2025-04-11:
Check that the Rsense value is correct (5160 is different from 2209) and that there is not too much current being pushed down to the motor. Too much current could result in overheating --> stalling --> undervoltage.

WillGIam7, 2025-04-11:
I was using sense_resistor: 0.022 and tried reducing the motor current down to 0,75 and even lower. Is there a way that I destroyed my drivers by forgetting the sense_resistor: 0.022 at the beginning? I´m not sure, but it may be. I was pairing the drivers with 2209 UART Mode on the same board. Could that be an issue? I already tried my octopus, which was running fine with these drivers and just added tmc2209 on the other slots and upgraded klipper.

bigtreetech, 2025-04-11:
> I was using sense_resistor: 0.022 and tried reducing the motor current down to 0,75 and even lower. Is there a way that I destroyed my drivers by forgetting the sense_resistor: 0.022 at the [...]

The sense resistor value tells the driver what voltage to target in order to meet the required current. Setting a value that is too high will result in far too much current being driven. For example, if the actual sense resistor is 0.022R and you set it to 0.075R then you will see more than 3x the requested current being driven into the load in order to reach the expected voltage. This may indeed cause damage to the FETs on the external driver or the motor itself or even the PSU. It really depends on how long it was done for.

WillGIam7, 2025-04-11:
> The sense resistor value tells the driver what voltage to target in order to meet the required current. Setting a value that is too high will result in far too much current being driven. For [...]

Thanks for your answers. Im very happy getting this fast replies. 

So I tested some other drivers with the right Rsense and they dont work too. It seems like there is another problem.

I´m testing through the troubleshooting from this thread (https://github.com/Klipper3d/klipper/issues/2516) now, but I really dont get the problem. 

I now checked the PSU, I used another 24V PSU instead of the 48V: Same error. The 48V are stable, but I didnt checked that with an Oszi. Since the error occurs in both case I think the PSU is fine. 

I also changed the board form a M8P to a octopus: Same error. The board seems fine. 2209 are running fine.

I then changed the drivers and still getting the problem. Based on that not all my 7 drivers are defect, the error seems to be elsewhere. To clarify: These are TMC5160T Plus (maybe an older revision?) with and without case. On the other hand I got the drivers running on that octopus a few months ago with this psu and these motors. But I updated Klipper and added 2209er drivers. So I also tried the setup with only TMC5160T and: Problem is still there. 

And last but not least I changed the cables and the motors: Same Problem.

Now there is the possbility, that I killed all drivers with wrong settings (even with Rsense active for the last drivers) in any way. Second thing is, that I got the drives moving for a few seconds on both boards, after that I got this error and no moving anymore.

I added my config with the inputs of the named reddit thread. Maybe you have some ideas from that. 

Is there a way to send the drivers to you @bigtreetech and get a check or replacement if it is not my fault?

WillGIam7, 2025-04-11:
Hi, 

I have kind of a conclusion: I got one driver working on my octopus and swapped all drivers through (again). The conclusion is, that 5 of 7 drivers are faulty now. This said I switched back to the manta with 2 working drivers. I got them both running with the exact same setup but after some seconds of moving driver x got the global scaler error. This one I also got in another of the defect drivers. The others are throwing the undervoltage error. So it looks like the manta is destroying the 5160 which is what Ive seen in some other posts as well. The defect drivers are also getting very hot. I´ll now ban this board from my house, even that I love the CB2 and CAN connection. 

I now have spent a lot of money for that conclusion, even that I didnt make a mistake. They were working at least for a few seconds, just like when I started this whole process.