bigtreetech/BIGTREETECH-OCTOPUS-Pro #41: Octopus Pro & MicroProbe V2.0

motoyasu-atsumi, 2023-09-26:
I am struggling with the combination of Octopus Pro & MicroProbe V2.0. 
Is there anyone out there who has gotten it to work well?

SplendidGoT, 2023-12-06:
Same here, just can't get it to work. It will test eject the probe a couple of times when I boot up the printer, but when trying to do a bed leveling the probe does nothing and the printer will shutdown. Contacted Bigtreetech but they never respond.

carhifitrier, 2023-12-24:
Same here, but just the short warning kicking in when probe is connected at startup. An AliExpress item description is saying, its not compatible with octopus pro 446..

motoyasu-atsumi, 2023-12-24:
The BTT website stated that Octopus Pro is not supported.

carhifitrier, 2023-12-25:
> The BTT website stated that Octopus Pro is not supported.

Could you please point us to that statement? I can not find that information on any BTT/Biqu site.

motoyasu-atsumi, 2023-12-25:
When you look at it, it's gone!
Here is a screenshot of what I saw.

[image]

carhifitrier, 2023-12-31:
Just if someone is having the same issue and wants to use the microprobe in combination with his Octopus pro board on klipper, i took the following workaround:

Get a RP2040 board ( some arduino/clones should still work too, i bought a waveshare rp2040 zero)
Just flash with klipper and connect it to the host as secondary mcu
Use 5v, gnd and 2 pins (i went with gpio14 and gpio15) of that board for the probe and add the needed config to printer.cfg.

That way one can use it without issues caused by that incompatibility.

If using the microprobe V2:
The internal Pullup of the rp2040 seems to work without issues, if it gives false positive triggers, you might have to add an extra 10k resistor between the probe pin and 5v.

adamster1983, 2024-01-28:
It does work got it working on my octopus pro 1.1 h723 have u set stepper z pg10 endtop pin  with  probe:z_virtual_endstop

Quard, 2024-02-09:
Octopus Pro has a PROBE dedicated socket that works fine with MicroProbe V2.

My wiring looks like:
- servo connector with 3 pins connected to the BLTouch socket, but just 3 pins(GND, 5V, Servo - PB6)
- and probe connector connected to the PROBE socket, just GND and Probe(PC5) pins, without DC
- connect jumper near the PROBE socket to enable pull-up resistor on Probe pin

and configuration for the Klipper
```
[output_pin probe_enable]
pin: PB6
value: 0

[gcode_macro Probe_Deploy]
gcode:
  SET_PIN PIN=probe_enable VALUE=1

[gcode_macro Probe_Stow]
gcode:
  SET_PIN PIN=probe_enable VALUE=0

[probe]
pin: !PC5
deactivate_on_each_sample: False
x_offset: -12.6
y_offset: -29.5
#z_offset: 1.0
speed: 5.0
activate_gcode:
  Probe_Deploy
  G4 P500
deactivate_gcode:
  Probe_Stow
```

carhifitrier, 2024-02-09:
Which revision/version of Octopus pro do you have it working with? I have an 1.0 F446, which is kicking in the short protection as soon as i fully have the microprobe connected. I might try your connection setup if you are using the same board sucessfully.

Quard, 2024-02-09:
Octopus Pro V1.1 with H723

adamster1983, 2024-02-10:
> According to the schematics in this repository Probe socket has the same
> implementation on both versions of the board. I means that here should not
> be any difference between the board versions for the MicroProbe.
>
> @carhifitrier <https://github.com/carhifitrier> could you show your
> wiring and configuration for the MicroProbe. For now, I have only one
> suggestion, you have enabled pull-up resistor in configuration, as it done
> in MicroProbe manual for all boards, but you should not do that as board
> already has the physical pull-down resistor on this PC5 pin(Probe).

Many thanks got it working much appreciated.

Thanks

Adam

carhifitrier, 2024-02-10:
@Quard 
Thanks for the Idea.
I just tried with connection of all 5 on the normally correct bltouch pins, will try with the 2 on probe port as soon as i get to it. My code is working, that's not a problem. 
Seems as O. pro 1.1 does work with it, did not hear about an user who did it with an 1.0 pro Board. 
I'll give feedback when i got into it!
Setting a second pulldown should rather be not leading to the issue i am experiencing.

carhifitrier, 2024-08-11:
Got back to this Issue when preparing one of my printers for sale to a workmate. To simplify, i tried solving this with a serial resistor, limiting current to the probe when switching on. 10 Ohm were too much, but 1 Ohm at VCC works perfect. 
So the hardware way to use a Octopus pro 1.0 with microprobe V2.0 is a 1 Ohm resistor. Overcurrent / Short protection does not kick in anymore and the Probe just works as intended, connected at BLT connector, PB6(en) and PB7 :)

casareanderson, 2025-05-30:
@Quard  do you have a image of how you got it wired i am a little confused i have the H723 and trying to get it to work i have followed your steps but would be amazing to see a  picture of how you have it wired

Quard, 2025-06-09:
@casareanderson unfortunately I do not have access to the printer for next 2-3 month