bigtreetech/BIGTREETECH-OCTOPUS-Pro #48: Octopus pro 723

jimgineer, 2024-05-19:
The Octopus pro V1.1 board with the STM32FH723ZE chip is lacking with clear information. I feel like in an effort to save time any information on that particular board was vaguely put in the margins of manuals with a similar layout.  To me it screams lazy and does not make that part of the build process possible to move forward confidently without stress of ruining the board.

PZrk73, 2024-05-19:
Unfortunately, these are the realities of this time. The authors are generally reluctant to support the project. There is a circuit for the product. This is no small thing! I'm trying to figure it out by understanding the circuit.

Manu512, 2024-10-22:
I agree, clear documentation is essential for customers and DIY enthusiasts. If you're looking for specific information

bigtreetech, 2024-10-22:
The different MCUs share the same pin numbers aside from those mentioned in the readme. The documentation is equally applicable to both.

closed 2024-10-22 by bigtreetech