bigtreetech/BIGTREETECH-OCTOPUS-Pro #50: 12V rail - wrong current rating in manual?

NyxCode, 2024-08-27:
The manual states
> Take note that **the 12V rail can supply up to 4A**, the 5V up to 8A and the 3.3V
up to 1A however, in order to prevent overloading the rails it is not recommended to load
them to their maximum capacity since there are already various components on the
motherboard which also take their supply from them.

However, looking at the schematic, it seems like a [AOZ1283PI](https://www.digikey.com/en/products/detail/alpha-omega-semiconductor-inc/AOZ1283PI/3973646) is as a buck converter, which is only rated at 2.5A.

Either I'm missing something, or that information is false and dangerous.

Sigi-cz, 2025-04-11:
you are correct, I just went here to report the same thing. The IC can be damage at 3A already (Peak Current Limit 3A Min)

bigtreetech, 2025-04-11:
The peak current limit is rated at 4A typical in the datasheet. Continuous draw is rated at 2.5A. If a user exceeds the peak level for too long then the IC has over current, short circuit and over temperature protection integrated.

[image]

closed 2025-04-11 by bigtreetech

Sigi-cz, 2025-04-11:
AMS1117 used to generate 3V3, you report correctly nominal 1A, current limit is 2.1A.
SY8368A used to generate 5V, you report correctly nominal 8A, current limit is 16A.
AOZ1283 used to generate 12V has nominal 2.5A, current limit is 3A. Some IC produced can be damaged at 3A !
Why are you lying about the 12V rail?