bigtreetech/BIGTREETECH-OCTOPUS-Pro #52: Missing Bootloaders

EmperorArthur, 2025-02-23:
# Issue
This repository is missing the bootloaders for the different board variants.

# Partial Solution
Attached is the bootloader for the BTT Octopus Pro V1.1 STM32H723ZET6 board.

[BTT Octopus Pro V1.1 STM32H723ZET6 Bootloader (Dumped 2025-02-23).zip](https://github.com/user-attachments/files/18932427/BTT.Octopus.Pro.V1.1.STM32H723ZET6.Bootloader.Dumped.2025-02-23.zip)

bigtreetech, 2025-02-23:
All identical to the standard octopus found here with instructions. https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/tree/master/Firmware/DFU%20Update%20bootloader/bootloader

closed 2025-02-23 by bigtreetech

EmperorArthur, 2025-02-24:
Thank you.  Would it be possible to add a note to look in the other repository to the README?  I am likely not the only person who is used to bootloaders changing with every board revision.

bigtreetech, 2025-02-25:
Yea we can do that.