bigtreetech/BIGTREETECH-OCTOPUS-Pro #53: BTT UPS 24V and BTT SFS V2.0 Module not work

Desktop-Solution, 2025-06-21:
Hello everyone,

Does anyone know how to make BTT UPS 24V and BTT SFS V2.0 Module work on the OCTOPUS Pro v1.1 board?
I downloaded the latest version of Marlin 2.1.2.5, I also tried the one from bigtreetech that he uploaded to github, but nothing works.

I made several attempts to configure and connect on the board, I even changed the port pins but they don't work.

Basically BTT UPS 24V, the loss of current is not detected and therefore the Z axis does not rise.

CONFIG:

`#define POWER_LOSS_RECOVERY
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define PLR_ENABLED_DEFAULT   true // Power Loss Recovery enabled by default. (Set with 'M413 Sn' & M500)
    #define BACKUP_POWER_SUPPLY       // Backup power / UPS to move the steppers on power loss
    #define POWER_LOSS_ZRAISE       5 // (mm) Z axis raise on resume (on power loss with UPS)
    #define POWER_LOSS_PIN        PC0 // Pin to detect power loss. Set to -1 to disable default pin on boards without module.
    #define POWER_LOSS_STATE     HIGH // State of pin indicating power loss
    #define POWER_LOSS_PULLUP         // Set pullup / pulldown as appropriate for your sensor
    //#define POWER_LOSS_PULLDOWN
    #define POWER_LOSS_PURGE_LEN   20 // (mm) Length of filament to purge on resume
    #define POWER_LOSS_RETRACT_LEN 10 // (mm) Length of filament to retract on fail. Requires backup power.

    // Without a POWER_LOSS_PIN the following option helps reduce wear on the SD card,
    // especially with "vase mode" printing. Set too high and vases cannot be continued.
    #define POWER_LOSS_MIN_Z_CHANGE 0.05 // (mm) Minimum Z change before saving power-loss data

    // Enable if Z homing is needed for proper recovery. 99.9% of the time this should be disabled!
    //#define POWER_LOSS_RECOVER_ZHOME
    #if ENABLED(POWER_LOSS_RECOVER_ZHOME)
      //#define POWER_LOSS_ZHOME_POS { 0, 0 } // Safe XY position to home Z while avoiding objects on the bed
    #endif
  #endif`

For BTT SFS V2.0 powered correctly (the LEDs show a positive feedback of when there is filament (blue) and when there is not (red)) the print does not pause. I also measured the voltage with a multimeter, (3.35 volts when the filament is present, 0 when it is pulled away). While for the movement it sends a variable voltage in volts to the board it is deaf!

CONFIG:

`#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.

  #define FIL_RUNOUT_STATE     LOW        // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN           // Use internal pulldown for filament runout pins.
  //#define WATCH_ALL_RUNOUT_SENSORS      // Execute runout script on any triggering sensor, not only for the active extruder.
                                          // This is automatically enabled for MIXING_EXTRUDERs.

  // Override individually if the runout sensors vary
  //#define FIL_RUNOUT1_STATE LOW
  //#define FIL_RUNOUT1_PULLUP
  //#define FIL_RUNOUT1_PULLDOWN

  //#define FIL_RUNOUT2_STATE LOW
  //#define FIL_RUNOUT2_PULLUP
  //#define FIL_RUNOUT2_PULLDOWN

  //#define FIL_RUNOUT3_STATE LOW
  //#define FIL_RUNOUT3_PULLUP
  //#define FIL_RUNOUT3_PULLDOWN

  //#define FIL_RUNOUT4_STATE LOW
  //#define FIL_RUNOUT4_PULLUP
  //#define FIL_RUNOUT4_PULLDOWN

  //#define FIL_RUNOUT5_STATE LOW
  //#define FIL_RUNOUT5_PULLUP
  //#define FIL_RUNOUT5_PULLDOWN

  //#define FIL_RUNOUT6_STATE LOW
  //#define FIL_RUNOUT6_PULLUP
  //#define FIL_RUNOUT6_PULLDOWN

  //#define FIL_RUNOUT7_STATE LOW
  //#define FIL_RUNOUT7_PULLUP
  //#define FIL_RUNOUT7_PULLDOWN

  //#define FIL_RUNOUT8_STATE LOW
  //#define FIL_RUNOUT8_PULLUP
  //#define FIL_RUNOUT8_PULLDOWN

  // Commands to execute on filament runout.
  // With multiple runout sensors use the %c placeholder for the current tool in commands (e.g., "M600 T%c")
  // NOTE: After 'M412 H1' the host handles filament runout and this script does not apply.
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  #define FILAMENT_RUNOUT_DISTANCE_MM 3

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    #define FILAMENT_MOTION_SENSOR
  #endif
#endif`

PIN PAGE:

`#define FIL_RUNOUT_PIN                      PG12  // E0DET
#define FIL_RUNOUT2_PIN                     OFF   // PG13  // E1DET
#define FIL_RUNOUT3_PIN                     PG14
#define FIL_RUNOUT4_PIN                     PG15
#define FIL_MOTION1_PIN                     PG13`

Is anyone kind enough to help me?
Thanks
Luca

usama0300854, 2025-12-11:
Im facing the same issue, octopus pro saves the plr file but doesnt ask for resume print when printer restart. 

Everything works best on ramps board with arduino mega.

Im think of switching to klipper for exact reading of print failing.

usama0300854, 2025-12-11:
Did you find any solution?

usama0300854, 2025-12-16:
I know every printer is different but i tried everything. I bought two of your boards. Octopus v1.1 and octopus pro v1.0.

I just want to say that i have tried my best. I build custom printers so trust me i tried!

Anyways the plr file does get created in the sd card but it doesnot give print resume option!

I think the sd card is mounted later in the firmware because you have to go in menu to attach the sd card and then it shows up!

usama0300854, 2026-01-02:
I solved the issue.
Finally power loss recovery is working and resume print is showing up on lcd when printer restarts!!!

usama0300854, 2026-01-02:
Haha i changed few things in the marlin firmware. I made changes in this file : 

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/MarlinCore.cpp

Desktop-Solution, 2026-01-02:
> Haha i changed few things in the marlin firmware. I made changes in this file :
> 
> https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/MarlinCore.cpp

Ok, can you attach the modified file? or explain where the problem lies. Thank you

usama0300854, 2026-01-02:
Through fiverr ofcourse.

I will upload a video on YouTube so you can see its working.